periph/host #1: possible porting allwinner H3 or orangepi pc?

chenjeff622, 2021-01-01:
as title. how do i porting please?

maruel, 2021-01-01:
Could you contribute one?

If you want the headers, you could create a directory orangepi and use https://github.com/periph/host/tree/main/pine64 as an inspiration, then send a PR.

chenjeff622, 2021-01-08:
okey. i will give a try.

f0086, 2022-09-26:
@chenjeff622 Any progress on this? If not, I could pick this one up.
I want to support the Orange Pi Zero (v1) with an Allwinner H2 CPU. Sadly, the ```allwinner``` package does not detect the CPU (see here the [detection logic](https://github.com/periph/host/blob/main/allwinner/detect.go#L71-L81)). The H2 is a 32 bit CPU with the device string ```allwinner,sun8i-h2-plus```. Can I just extend the check for this type to support the CPU or is there more work to be done? What do you think @maruel?

chenjeff622, 2022-09-26:
https://github.com/periph/host/tree/main/allwinner
You should to create h2 inside the folder. I don't have time to test my h3 board. Maybe we make it success together.

maruel, 2022-09-26:
Even if the pinout is not supported, you can still use sysfs driver as a fallback.
I'm happy to have more memory registered driver! I don't have one to develop and test on.

PR #23 (2022-12-16): allwinner: Support for H2+ and H3 (OrangePiZero)
Add support for the Allwinner H2+ and H3 CPUs, which are almost identical. The H2+ has no GBit MAC and no 4h HDMI, but the rest is completely identical. For this library tho, it is not relevant, so we can combine these two CPU types into one single H3 and detect the H2+ as a H3 CPU.

Support for detecting Orange Pi Zero boards are added, which uses the H3 SOC. The Pinouts are from the Orange Pi manual.

Fixes #1
commit 339d73cb (2022-12-16): allwinner: Support for H2+ and H3 (OrangePiZero) (#23)

* Support for Allwinner H2+ and H3 (OrangePiSZero)

Add support for the Allwinner H2+ and H3 CPUs, which are almost
identical. The H2+ has no GBit MAC and no 4h HDMI, but the rest is
completely identical. For this library tho, it is not relevant, so we
can combine these two CPU types into one single H3 and detect the H2+ as
a H3 CPU.

Support for detecting Orange Pi Zero boards are added, which uses the H3
SOC. The Pinouts are from the Orange Pi manual.

Fixes #1
closed 2022-12-16 by maruel