periph/host #2: pmem example code incorrect

martin2250, 2021-05-12:
I think there is an error in the pmem example for MapAsPOD.

```go
package main

import (
	"log"

	"periph.io/x/periph/host/pmem"
)

func main() {
	// Let's say the CPU has 4 x 32 bits memory mapped registers at the address
	// 0xDEADBEEF.
	var reg *[4]uint32
	if err := pmem.MapAsPOD(0xDEADBEAF, reg); err != nil {
		log.Fatal(err)
	}
	// reg now points to physical memory.
}
```

It only works when I add a referencing operator to the function call, like so: `pmem.MapAsPOD(0xDEADBEAF, &reg)`, so this code works for me:

```go
var reg *[SMC_MEM_LEN / 2]uint16
if err := pmem.MapAsPOD(SMC_MEM_START, &reg); err != nil {
```

Otherwise, I get an error during runtime: `pmem: require Ptr to Ptr, got invalid` (or `got slice`/`got array`).

In case this is relevant, I'm compiling for an AT91SAM9G45 with Linux 3.0.0.

maruel, 2021-05-12:
Small note; `"periph.io/x/host/v3/pmem"`

martin2250, 2021-05-12:
right, I updated the code snippet.

kaisawind, 2021-12-27:
same.
Linux 5.4.164-1-MANJARO #1 SMP PREEMPT Wed Dec 8 09:53:07 UTC 2021 x86_64 GNU/Linux

PR #66 (2025-01-17): pmem: fix example
Fixes #2
closed 2025-01-17 by maruel