periph/host #7: How do I Open a Serial Port?

gsexton, 2021-08-01:
I'm trying to figure out how to open a serial port, but not having a lot of luck.

The only thing that looks right is periph/hosts/serial/serial_go, but (leaving aside the port naming ttyS? idea is broken), func newPortDevFs(portNumber int) (*Port, error) doesn't seem to be exported, so I'm at a loss.

Can you give me an idea on how to you instantiate a serial port?

Thanks.

George

maruel, 2021-08-01:
Hi!

Serial support needs to be implemented. Initially periph was not importing any other packages so the initial expectation was to implement it with raw API calls. Starting with periph v3 it's now fine to reuse previous implementations so that could be a way to go to have something up and running more quickly. My preference would be to not depend on cgo but it may be fine in practice. Would you want to work on such feature?

gsexton, 2021-08-10:
Thanks. I'll look at it after I get my current project done. I'm pretty new to golang, so I'm not sure how well I can do.

commit 6113c6e0 (2020-12-24): doc: add code style, note about lack of affiliation with Go. (#10)

Fixes #7
closed 2021-08-10 by gsexton