periph/host #17: ftdi depends on cgo

schmidtw, 2022-10-01:
**Describe the bug**
The host package depends on periph/d2xx which depends on cgo.
https://github.com/periph/d2xx/blob/86eb5ae2a9f0cabc315eff6641cd8aea22df5b6c/d2xx_linux_amd64.go#L5

This dependency should probably be fixed or removed to maintain the [No C dependency, doesn’t use cgo.](https://periph.io/) policy.

**To Reproduce**
1. Install go compiler on a fresh RPI install.  Do not install c compiler.
2. `git clone https://github.com/periph/cmd.git`
3. `cd cmd`
4. `go build ./...`
5. See error:
```
# periph.io/x/d2xx
cgo: C compiler "gcc" not found: exec: "gcc": executable file not found in $PATH
```

**Expected behavior**
Build succeeds.

**Platform (please complete the following information):**
 - OS: RockyLinux
 - Board RPI4

**Additional context**
The ftdi libraries are neat, but probably shouldn't be in the periph projects unless you segment out "here are projects that are related but use cgo".

maruel, 2022-10-01:
d2xx depends on cgo only if cgo is enabled. Set the environment variable `CGO_ENABLED=0` to not depend on the cgo enabled code. Is this good enough for you?

schmidtw, 2022-10-01:
Yes, I missed that I could do that.  Thank you.

closed 2022-10-01 by schmidtw