periph/host #18: Serial Interface Update Question

schmidtw, 2022-10-02:
The serial port sub-system was [mentioned](https://github.com/periph/host/issues/7) as one that you're ok bringing in other implementations.

I'm presently using this one as the basis for a ds2480 1-wire chip: https://github.com/bugst/go-serial.  Based on my searching this seems to be the best cross-platform, no cgo version around.

I'm curious how you envision supporting other impls.  I think the `conn.Conn` interface works for most simple serial devices, but the `Connect()` function may not be enough to support devices like a ds2480, where uart focused changes need to happen on the fly.

If there is a set of interfaces that are the right ones for implementing as external serial port adapters, then it will be pretty easy to write a small adapter library.  The adapter can be external to the host project (like `periph/bugst-serial-adapter`) that can adapt the two & allow for the drivers & that logic to live in periph/drivers.

I'm happy to do the coding for these two things (ds2480 and the bugst/go-serial bridge) as long as I know what you want it to look like & you're cool with the approach.

maruel, 2022-10-03:
You mention Connect, you mean https://pkg.go.dev/periph.io/x/conn/v3/uart#Port ? I'm confused because you refer to 1-wire but the 1-wire API doesn't have a Connect() function; https://pkg.go.dev/periph.io/x/conn/v3/onewire

This is because UART is a Port, and 1-wire is a Bus.

maruel, 2022-10-04:
Oh ok so it would work over UART instead of I2C like https://pkg.go.dev/periph.io/x/devices/v3/ds248x does?

schmidtw, 2022-12-17:
Yes, that's right.  Sorry for the delay - I missed the update email & just saw your post.