periph/host #29: GPIO unexport at exit

sausagenoods, 2023-02-23:
**Describe the bug**
I need to manually "unexport" pins after using this library. Let's say I've used pin 17 with the [debounce example code](https://pkg.go.dev/periph.io/x/conn/v3/gpio/gpioutil#example-Debounce). I'm unable to use the same pin again ([using another library](https://github.com/warthog618/gpiod)) without rebooting the Pi or running a command like this:
```sh
$ sudo bash -c "for ((i=0; i<32; i++)); do echo \$i >/sys/class/gpio/unexport; done"
```
Otherwise the other library fails with: `RequestLine returned error: device or resource busy`. And no I'm not running both programs at the same time. Apologies if this is normal, I don't know much about GPIO.
**To Reproduce**
Steps to reproduce the behavior:
1. Run the [debounce example code](https://pkg.go.dev/periph.io/x/conn/v3/gpio/gpioutil#example-Debounce).
2. Run the [input pin watcher example from the other library](https://github.com/warthog618/gpiod/blob/master/example/select_watcher/select_watcher.go).
3. 2nd step fails with the RequestLine error I mentioned above and I'm unable to reuse the same pin unless I manually unexport it.

**Expected behavior**
Be able to reuse the same pin. I think it would be nice to have a function that unexports and people would defer that.

**Platform (please complete the following information):**
 - OS: Raspberry Pi OS Bullseye 5.15.84-v8+
 - Board: Raspberry Pi 3b+

maruel, 2023-02-23:
We would need a new shutdown function exported from either sysfs or as a generic form in periph.

gsexton, 2025-01-13:
This should no longer be required with the ioctl() gpio access.

commit 8bf69817 (2020-12-24): fix i2c read-only Tx, fixes #29 (#30)
closed 2025-01-13 by gsexton