periph/host #35: Drivers skipped

ea7kir, 2023-07-20:
Can't get you example code to work.

**To Reproduce**
Copy your code into VsCode.
```
package main

import (
	"fmt"
	"log"

	"periph.io/x/host/v3"
)

func main() {
	// Make sure periph is initialized.
	state, err := host.Init()
	if err != nil {
		log.Fatalf("failed to initialize periph: %v", err)
	}

	// Prints the loaded driver.
	fmt.Printf("Using drivers:\n")
	for _, driver := range state.Loaded {
		fmt.Printf("- %s\n", driver)
	}

	// Prints the driver that were skipped as irrelevant on the platform.
	fmt.Printf("Drivers skipped:\n")
	for _, failure := range state.Skipped {
		fmt.Printf("- %s: %s\n", failure.D, failure.Err)
	}

	// Having drivers failing to load may not require process termination. It
	// is possible to continue to run in partial failure mode.
	fmt.Printf("Drivers failed to load:\n")
	for _, failure := range state.Failed {
		fmt.Printf("- %s: %v\n", failure.D, failure.Err)
	}

	// Use pins, buses, devices, etc.
}
```
and then...
```
go mod init example
go mod tidy
```
2. Run it.
```
go run .
```
3. See error
```
Using drivers:
- bcm283x-gpio
- rpi
- sysfs-gpio
- sysfs-i2c
- sysfs-led
- sysfs-thermal
Drivers skipped:
- allwinner-gpio-pl: no A64 CPU detected
- allwinner-gpio: no Allwinner CPU detected
- pine64: pine64 board not detected
- sysfs-spi: no SPI port found
Drivers failed to load:
- bcm283x-dma: pmem: failed to open physical memory: open /dev/mem: permission denied
```
**Expected behavior**
No Drivers skipped and no Drivers failed to load.

## Platform
 - OS: Pi OS (64bit)
 - Board: Raspberry Pi 4

**Additional context**
This is after 3 days of trying to find proper documentation on how to install and get started.  Please help.

maruel, 2023-07-20:
Drivers that are not applicable are skipped.

Drivers that fail because the program is not running as root correctly fail to load. This is normal behavior.

maruel, 2023-07-20:
It's okay. You don't need the bcm283x-dma driver to use periph. You can go ahead and write code.

closed 2023-07-20 by maruel