periph/host #44: bcm283x: revamp DMA support

maruel, 2023-09-20:
Ported from https://github.com/google/periph/issues/172.

This is about cleaning up the RPi DMA code for all generations.

Also see https://github.com/periph/host/issues/34 for RPi4 specific issue.

maruel, 2023-09-20:
Also including issue https://github.com/google/periph/issues/198 about misuse of dmaWaitcyclesMax.

kirwinrMK, 2024-04-12:
DMA appears completely broken on RPI. 
line 916 in bcm283x fails due to a miss match in units:
`l := int(int64(d) * int64(w.Frequency()) / int64(physic.Hertz)) // Bits` should be `l := int(d / w.Frequency().Period()) // Bits'

Once that's fixed, there's still a SIGBUS error coming from Nanospin, that could be because I'm using a CM4 though.