periph/host #51: host.Init forked processes persist and maintain high CPU load

BrockMatthews, 2023-10-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
	"log"
	"runtime"
	"time"

	"periph.io/x/host/v3"
	"periph.io/x/host/v3/ftdi"
)

func main() {
	host.Init()
	log.Println("Init done.")

	ftdi.All()

	buffer := make([]byte, 1024)
	runtime.Stack(buffer, true)

	log.Println("Running goroutines:")
	log.Println(string(buffer))

	time.Sleep(1 * time.Hour)
}
```
2. Run it.
```
2023/10/04 00:05:43 Init done.
2023/10/04 00:05:43 Running goroutines:
2023/10/04 00:05:43 goroutine 1 [running]:
main.main()
	/home/name/go/src/periphio_test/test.go:19 +0x78
```
3. Check process
```
# top -H -p <test PID>

Threads:  12 total,   0 running,  12 sleeping,   0 stopped,   0 zombie
%Cpu(s):  4.7 us, 19.3 sy,  0.0 ni, 75.7 id,  0.0 wa,  0.0 hi,  0.3 si,  0.0 st
KiB Mem :  3074384 total,  2296660 free,   562180 used,   215544 buff/cache
KiB Swap:        0 total,        0 free,        0 used.  2406060 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND                                                                                                                                               
 5264 name      20   0 1327928  14104   3280 S 13.0  0.5   0:08.36 test                                                                                                                                       
 5278 name      20   0 1327928  14104   3280 S 12.6  0.5   0:08.72 test                                                                                                                                       
 5277 name      20   0 1327928  14104   3280 S 10.0  0.5   0:06.38 test                                                                                                                                       
 5262 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.01 test                                                                                                                                       
 5263 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.00 test                                                                                                                                       
 5265 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.00 test                                                                                                                                       
 5266 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.01 test                                                                                                                                       
 5267 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.00 test                                                                                                                                       
 5268 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.00 test                                                                                                                                       
 5269 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.00 test                                                                                                                                       
 5270 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.00 test                                                                                                                                       
 5279 name      20   0 1327928  14104   3280 S  0.0  0.5   0:00.00 test 
```

**Expected behavior**
No processes to persist and produce high CPU load after host.Init() completion. 

**Platform (please complete the following information):**
 - OS: NXP i.MX Release Distro 4.14-sumo (sumo)
 - Board: custom

**Additional context**
Device: FTDI FT232H

Device is completely usable after host.Init()

ftdi_sio driver unbound from used FT232H, but module not removed as it is required by other FTDI devices on the system.

maruel, 2023-10-04:
This is surprising. Can you use https://pkg.go.dev/runtime/pprof to try to determine what is spinning? Keep in mind the ftdi driver is known to not be at the quality bar that the project generally aims to achieve.

BrockMatthews, 2023-10-04:
Thanks. The profiling output isn't awfully specific but here it is
```
Duration: 10.21s, Total samples = 5.66s (55.44%)
Entering interactive mode (type "help" for commands, "o" for options)
(pprof) top
Showing nodes accounting for 5660ms, 100% of 5660ms total
      flat  flat%   sum%        cum   cum%
    2390ms 42.23% 42.23%     2390ms 42.23%  [libpthread-2.27.so]
    2330ms 41.17% 83.39%     2330ms 41.17%  runtime._LostExternalCode
     840ms 14.84% 98.23%      840ms 14.84%  [libc-2.27.so]
     100ms  1.77%   100%      100ms  1.77%  [PeriphioTest_aarch64]
         0     0%   100%     5660ms   100%  runtime._ExternalCode
```
Please let me know of anything else I can provide.

maruel, 2023-10-04:
Sorry I won't be able to diagnose with this data. You will likely need to investigate more deeply yourself.

BrockMatthews, 2023-10-05:
On deeper inspection the issue seems to be caused by the ftd2xx driver with the call to FT_Open in d2xx_posix.go open(). This happens specifically on the device that is used and has ftdi_sio unbound, and on none of the other FTDI devices iterated through.

closed 2025-01-13 by gsexton