periph/host #53: uart: Not detecting serial ports on RPi 3 or 4

adev73, 2024-05-13:
**Describe the bug**
Library is not able to locate any uarts on a Raspberry Pi (have tested a 3+ and a 4).

**To Reproduce**
Steps to reproduce the behavior:
Configure RPi so that /dev/serial0 (and /dev/serial1 on a 4) are available.
Configure disable-bt to make sure bluetooth is switched off.

Run the "ExampleAll" function from the uart device folder:

```go
package main

import (
	"fmt"
	"log"
	"strings"

	"periph.io/x/conn/v3/uart"
	"periph.io/x/conn/v3/uart/uartreg"
	"periph.io/x/host/v3"
)

func main() {
	// Make sure periph is initialized.
	if _, err := host.Init(); err != nil {
		log.Fatal(err)
	}

	// Enumerate all UART ports available and the corresponding pins.
	fmt.Print("UART ports available:\n")
	for _, ref := range uartreg.All() {
		fmt.Printf("- %s\n", ref.Name)
		if ref.Number != -1 {
			fmt.Printf("  %d\n", ref.Number)
		}
		if len(ref.Aliases) != 0 {
			fmt.Printf("  %s\n", strings.Join(ref.Aliases, " "))
		}

		b, err := ref.Open()
		if err != nil {
			fmt.Printf("  Failed to open: %v", err)
		}
		if p, ok := b.(uart.Pins); ok {
			fmt.Printf("  RX : %s", p.RX())
			fmt.Printf("  TX : %s", p.TX())
			fmt.Printf("  RTS: %s", p.RTS())
			fmt.Printf("  CTS: %s", p.CTS())
		}
		if err := b.Close(); err != nil {
			fmt.Printf("  Failed to close: %v", err)
		}
	}
}
```
2. Output:

```
pi@wmrm:~/source/wmrm $ go run main.go
UART ports available:
pi@wmrm:~/source/wmrm $ 
```

**Expected behavior**
There should have been a list of at least one available UART, preferably two

**Platform (please complete the following information):**
 - OS: Raspbian Bullseye Lite
 - Version: Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
 - Board: Raspberry Pi 3+ or 4
 - Go: Version 1.18.3

PaulAtST, 2024-05-13:
@maruel I noticed when you moved the issue to host from con, that host has a serial package. The serial package does seam to enumerate my device, but I don't see how I can create a serial port. It looks like I would need to call `func newPortDevFs(portNumber int) (*Port, error) {` which is unexported.

maruel, 2024-05-14:
Oh it's just that nobody has implemented the linux serial driver yet. The implementation belongs to host, not to conn, that's why I moved the issue.