periph/host #62: Edge detection does not work on Raspberry Pi 2

fischerman, 2024-09-23:
**Describe the bug**

The pin cannot be associated with a sysfs pin.

```
$ go run cmd/edge/edge.go 
2024/09/23 10:22:18 bcm283x-gpio (GPIO17): pin 17 is not exported by sysfs
exit status 1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run program
```go
package main

import (
	"log"

	"periph.io/x/conn/v3/gpio"
	"periph.io/x/conn/v3/gpio/gpioreg"
	"periph.io/x/host/v3"
)

func main() {
	if _, err := host.Init(); err != nil {
		log.Fatal(err)
	}

	// Lookup a pin by its number:
	p := gpioreg.ByName("GPIO17")
	if p == nil {
		log.Fatal("Failed to find GPIO17")
	}

	// Set it as input, with an internal pull down resistor:
	if err := p.In(gpio.PullDown, gpio.BothEdges); err != nil {
		log.Fatal(err)
	}

	p.WaitForEdge(-1)
}
```
2. See error

**Expected behavior**

No error on `In`.

**Platform:**
 - OS: Raspbian GNU/Linux 12 (bookworm)
 - Board: Raspberry Pi 2 Model B v1.1

**Additional context**

I'm not aware of any prerequisites that must be met. I tried running the program as root as well, but I get the same error.

fischerman, 2024-09-23:
I think the problem is that starting with Bookworm sysfs GPIO has been removed. It might affect other distributions as well. At least all Raspberries are affected.

- https://www.thegoodpenguin.co.uk/blog/stop-using-sys-class-gpio-its-deprecated/
- https://forum.fhem.de/index.php?topic=137661.0

maruel, 2024-09-23:
Can you test if https://github.com/periph/host/commit/b69b28c4e40f9f233222c9d7ea610ef0fcc8a54b fixes the problem? If so, I'll do a release.

fischerman, 2024-09-24:
On the Pi 2 I'm getting a lot of errors now:
```
go run cmd/edge/edge.go 
2024/09/24 10:28:43 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 33,
    "Name": "NC",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "NC" twice; already registered as "{\n    \"Line\": 30,\n    \"Name\": \"NC\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
2024/09/24 10:28:43 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 34,
    "Name": "NC",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "NC" twice; already registered as "{\n    \"Line\": 30,\n    \"Name\": \"NC\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
2024/09/24 10:28:43 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 36,
    "Name": "NC",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "NC" twice; already registered as "{\n    \"Line\": 30,\n    \"Name\": \"NC\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
2024/09/24 10:28:43 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 37,
    "Name": "NC",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "NC" twice; already registered as "{\n    \"Line\": 30,\n    \"Name\": \"NC\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
2024/09/24 10:28:43 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 39,
    "Name": "NC",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "NC" twice; already registered as "{\n    \"Line\": 30,\n    \"Name\": \"NC\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
2024/09/24 10:28:43 bcm283x-gpio (GPIO17): subsystem gpiomem not initialized and sysfs not accessible
exit status 1
```

Not sure if it is related but header-list no longer returns functions:
```
$ go run headers-list/main.go 
AUDIO: 2 pins
  Pos  Name    Func
  1    GPIO45  
  2    GPIO40  

HDMI: 1 pins
  Pos  Name    Func
  1    GPIO46  

P1: 40 pins
  Func    Name  Pos  Pos  Name    Func
          3.3V    1  2    5V          
         GPIO2    3  4    5V          
         GPIO3    5  6    GROUND      
         GPIO4    7  8    GPIO14      
        GROUND    9  10   GPIO15      
        GPIO17   11  12   GPIO18      
        GPIO27   13  14   GROUND      
        GPIO22   15  16   GPIO23      
          3.3V   17  18   GPIO24      
        GPIO10   19  20   GROUND      
         GPIO9   21  22   GPIO25      
        GPIO11   23  24   GPIO8       
        GROUND   25  26   GPIO7       
         GPIO0   27  28   GPIO1       
         GPIO5   29  30   GROUND      
         GPIO6   31  32   GPIO12      
        GPIO13   33  34   GROUND      
        GPIO19   35  36   GPIO16      
        GPIO26   37  38   GPIO20      
        GROUND   39  40   GPIO21
```

gsexton, 2024-09-27:
@fischerman 
Can you try running this example:

host/gpioioctl/example_test.go

Evert-Arends, 2024-10-08:
I want to see this fixed so I ran a gpio out on my RPI 5, using the branch you provided @maruel 

```shell
pi@worker-pi-1:~/GraftXL/cmd/run3 $ go get periph.io/x/host/v3@b69b28c4e40f9f233222c9d7ea610ef0fcc8a54b
go: downloading periph.io/x/host/v3 v3.8.3-0.20240918234808-b69b28c4e40f
```

which quickly died:
```shell
2024/10/08 20:13:09 chip gpiochip4  gpioreg.Register(line)  {
    "Line": 47,
    "Name": "2712_WAKE",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "2712_WAKE" twice; already registered as "{\n    \"Line\": 8,\n    \"Name\": \"2712_WAKE\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
Failed to set pin as output: bcm283x-gpio (GPIO17): subsystem gpiomem not initialized and sysfs not accessible
```

Using this example code for my motor that works on the RPI4:

```go
package main

import (
    "fmt"
    "time"
    "periph.io/x/conn/v3/gpio"
    "periph.io/x/host/v3"
    "periph.io/x/host/v3/rpi"
)

func main() {
    if _, err := host.Init(); err != nil {
        fmt.Println("Failed to initialize periph:", err)
        return
    }

    stepPin := rpi.P1_11 // Use GPIO17

    // Set the pin as an output
    if err := stepPin.Out(gpio.Low); err != nil {
        fmt.Println("Failed to set pin as output:", err)
        return
    }

    pulseCount := 800

    pulseDuration := 1000 * time.Microsecond

    // Generate pulses
    fmt.Printf("Sending %d pulses...\n", pulseCount)
    for i := 0; i < pulseCount; i++ {
        if err := stepPin.Out(gpio.High); err != nil {
            fmt.Println("Failed to set pin high:", err)
            return
        }

        time.Sleep(pulseDuration / 2)

        if err := stepPin.Out(gpio.Low); err != nil {
            fmt.Println("Failed to set pin low:", err)
            return
        }

        time.Sleep(pulseDuration / 2)
    }

    fmt.Println("Finished sending pulses.")
}
```

I hope this helps in someway!

gsexton, 2024-10-08:
@fischerman 

There are 4 GPIO chips on the Pi 5, and they all export a pin named "2712_WAKE". This causes the error you're seeing.

Take the return out of the host.Init() statement and see what it does.

Evert-Arends, 2024-10-09:
Did that, removed the return.

```
2024/10/09 19:10:11 chip gpiochip4  gpioreg.Register(line)  {
    "Line": 47,
    "Name": "2712_WAKE",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "2712_WAKE" twice; already registered as "{\n    \"Line\": 8,\n    \"Name\": \"2712_WAKE\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
Failed to set pin as output: bcm283x-gpio (GPIO23): subsystem gpiomem not initialized and sysfs not accessible

```

```
func main() {
        if _, err := host.Init(); err != nil {
                fmt.Println("Failed to initialize periph:", err)
                //        return
        }

        stepPin := rpi.P1_16 // 23

        // Set the pin as an output
        if err := stepPin.Out(gpio.Low); err != nil {
                fmt.Println("Failed to set pin as output:", err)
                return
        }

        pulseCount := 800

        pulseDuration := 1000 * time.Microsecond

        // Generate pulses
        fmt.Printf("Sending %d pulses...\n", pulseCount)
        for i := 0; i < pulseCount; i++ {
                if err := stepPin.Out(gpio.High); err != nil {
                        fmt.Println("Failed to set pin high:", err)
                        return
                }

                time.Sleep(pulseDuration / 2)

                if err := stepPin.Out(gpio.Low); err != nil {
                        fmt.Println("Failed to set pin low:", err)
                        return
                }

                time.Sleep(pulseDuration / 2)
        }

        fmt.Println("Finished sending pulses.")
}
```

fischerman, 2024-10-12:
@gsexton Your example works on my RPi2. I compared it to my program and what seems to break it is:
```
gpioreg.ByName("GPIO17")
```
`gpioioctl.Chips[0]` on the other hand works fine.

gsexton, 2024-10-14:
@fischerman I've looked into this.

The problem is the bcm283x driver is running after the gpioioctl register code. That driver explicitly unregisters the gpioioctl registered GPIO pins values and supplies non-working sysfs based pins, causing things to break (line 1395 of host/bcm283x/gpio.go).

@maruel do you have any ideas on the correct fix?

maruel, 2024-10-15:
bcm283x needs to condition on which driver is loaded and act accordingly.

damdo, 2024-10-23:
Is this then solved by https://github.com/periph/host/pull/63 ?
Should this be closed? Thanks!

gsexton, 2024-11-03:
@damdo Could you test this on your configuration using the current head?

@fischerman and @Evert-Arends  Could you test this and see if your issue is resolved?

fischerman, 2024-11-04:
I've tested the following program with commit `7540d26b4dc114207d0e5f7603b0f176bf5805dd`.
```golang
package main

import (
	"fmt"
	"log"
	"time"

	"periph.io/x/conn/v3/gpio"
	"periph.io/x/conn/v3/gpio/gpioreg"
	"periph.io/x/host/v3"
)

func main() {
	if _, err := host.Init(); err != nil {
		fmt.Println("Failed to initialize periph:", err)
		//        return
	}

	outPin := gpioreg.ByName("GPIO2")
	if outPin == nil {
		log.Fatal("Failed to find GPIO2")
	}
	inPin := gpioreg.ByName("GPIO10")
	if inPin == nil {
		log.Fatal("Failed to find GPIO10")
	}
	inPin.In(gpio.PullDown, gpio.RisingEdge)
	fmt.Println("inPin=", inPin)
	fmt.Println("outPin=", outPin)

	l := gpio.Low
	go func() {
		for {
			l = !l
			fmt.Println("setting outPin to ", l)
			outPin.Out(l)

			time.Sleep(1000 * time.Millisecond)
		}
	}()

	for {
		if inPin.WaitForEdge(0) {
			fmt.Println("Received edge")
		} else {
			fmt.Println("WaitForEdge() unblocked without receiving edge.")
		}
	}
}

```

It works. However, I still see a lot of warnings:
```
2024/11/04 20:12:24 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 0,
    "Name": "ID_SDA",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "ID_SDA" twice; already registered as "{\n    \"Line\": 0,\n    \"Name\": \"ID_SDA\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
2024/11/04 20:12:24 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 1,
    "Name": "ID_SCL",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
[... 397 lines cut ...]
2024/11/04 20:12:24 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 51,
    "Name": "SD_DATA1_R",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "SD_DATA1_R" twice; already registered as "{\n    \"Line\": 51,\n    \"Name\": \"SD_DATA1_R\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
2024/11/04 20:12:24 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 52,
    "Name": "SD_DATA2_R",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "SD_DATA2_R" twice; already registered as "{\n    \"Line\": 52,\n    \"Name\": \"SD_DATA2_R\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}"
2024/11/04 20:12:24 chip gpiochip0  gpioreg.Register(line)  {
    "Line": 53,
    "Name": "SD_DATA3_R",
    "Consumer": "",
    "Direction": "NotSet",
    "Pull": "PullNoChange",
    "Edges": "NoEdge"
}  returned  gpioreg: can't register pin "SD_DATA3_R" twice; already registered as "{\n    \"Line\": 53,\n    \"Name\": \"SD_DATA3_R\",\n    \"Consumer\": \"\",\n    \"Direction\": \"NotSet\",\n    \"Pull\": \"PullNoChange\",\n    \"Edges\": \"NoEdge\"\n}
```

maruel, 2024-11-11:
@fischerman These errors are not supported to be there, that's weird.

gsexton, 2024-11-14:
@maruel 

I guess we need to discuss how to handle this. There's a lot going on.

Today I noticed that in a pi5 I have running, there's a symlink for /dev/gpiochip4 -> /dev/gpiochip0. That's probably causing most of the duplication.

The pi5 exports 5 total GPIO chips. Some of the lines have duplicate names. I see two chips (not counting the symlink) that have 2712_WAKE lines. So, we can't depend upon the exported line/pin names to be unique.

Looking at the output of the chips dumped (now you know why I did went wild on my String() implementation :) ). I can see that perhaps the parser should deterministically make a chip with label "pinctrl" the first one to be processed. Some of the chip labels appear to be pointers. For example:  "Label": "gpio-brcmstb@107d508500" btw.

So, our options are:

1) Don't print the errors. Programs break in mysterious ways because it's not going to be deterministic. The first release of bookworm for Pi5 was returning the chips in a different order than they are now. The effect will be that duplicates won't be accessible.

2) If the chip name looks like a pointer, assume it's not exposed and don't do anything with it. 

3) Keep doing all of the chips, but don't add colliding line names to the registry; just drop them. Probably not a bad choice.

4) Change the driver code so that it checks for the key. The bcm283x code is already doing this. We have two choices, if the key is duplicated, insert some sort of prefix (chip label) into the key and stick it in anyhow. This impacts the utility of the registry because you're not really going to be deterministic again. We can mitigate that somewhat by processing a chip labeled "pinctrl" first. There's also nothing that says the "pointer labels" will be the same across kernel updates. Sort of a vote for #2 above...

Alternatively, we can stick everything that is not labeled pinctrl into the registry with as label+line.

--

It's really sort of a mess. 

My recommendation is Option 2, and add code to handle duplicate chips caused by symlinks. Examine the /dev/ entry and skip if if it's a symlink. This is unfortunately a  Pi specific answer.

For a non-pi specific answer, my recommendation is: 1) Do a chip named "pinctrl-" first if found, and the rest in lexical order of device name. Skip symlinks. If a name is a collision, prefix it with the chip label.

I've attached a copy of the chip output from a pi5 running the lastest kernel if you'd like to look at it.

[chips.json](https://github.com/user-attachments/files/17754279/chips.json)

Let me know what you would like and I'll create another PR to implement it.

maruel, 2024-11-17:
Option 2 sounds fine with me. Thanks!

closed 2024-11-24 by maruel